PrusaOwners/klipper #30: "Lost communication with MCU" mid-print (after having successfully completed other prints)

ShanDestromp, 2019-05-09:
Not sure if this should technically be upstream at KevinOConnors end but I'm on a Prusa i3MK3 sourcing from this repo so I'll start here.

Printer had been running fine other than filament sensor not working (well it "works" in that you can poll it in Octoprint but doesn't pause on runout even though my printer.cfg has the right sections);  Saw recent commits to the tree so I updated.  Completed several prints yesterday including a 7+ hour print overnight.

Setup a new file to run this morning and it's now failed twice after a little over an hour with Octoprint stating 

> Your printer's firmware has reported an error.  Due to that the ongoing print job will be cancelled.  Reported error: Lost communication with MCU 'mcu'

It's failing *around* the same place but not exactly so I doubt it's a problem with the STL or sliced gcode
[image]

Top piece is first attempt, bottom is most recent

Attached is the klippy.log starting with the rollover at midnight tonight; it would have completed a print after the rollover, and then the two failed.  Also attached is both the STL and sliced gcode I'm using as well as my printer.cfg(.txt) just in case

[klippy.log](https://github.com/PrusaOwners/klipper/files/3162913/klippy.log)
[ER-16 Collet Holder.zip](https://github.com/PrusaOwners/klipper/files/3162928/ER-16.Collet.Holder.zip)
[printer.cfg.txt](https://github.com/PrusaOwners/klipper/files/3162943/printer.cfg.txt)

Arksine, 2019-05-09:
Your problem is the firmware on the Atmega32u2, it corrupts data over time leading to USB disconnections.  See [this repo](https://github.com/PrusaOwners/mk3-32u2-firmware) for more information.  Alternatively you can connect directly to the Einsy's Pi header to get around this issue.

ShanDestromp, 2019-05-09:
Is there something about my setup that makes it a sudden issue or just luck of the draw that it happened twice in quick succession?  I ask because this is the first time this has ever happened and I've been running Klipper/Octopi since ~Jan and that includes running some long prints (30+hours at 0.05mm layers); and before that on Prusa's firmware I was exclusively Octopi over USB there as well.

Directly connecting a Pi3 to the headers is less than ideal for a variety of reasons and updating the ATMega FW will have to wait until I've got more time (and a working Uno) to mess with.

Arksine, 2019-05-09:
I can only guess as to the actual cause as Prusa doesn't release the source for the 32u2, however I'm confident that it isn't dependent on your setup.  My guess is that something about the object you sliced is sending data in a way that corrupts data faster leading to a disconnection

ShanDestromp, 2019-05-10:
Based on your line of thought I reoriented and sliced with no significant change (failed at roughly the same spot as previous two attempts); I then went back and re-exported the STL with a reduced polygon count (roughly half actually).  No actual changes to the model, just the exported STL.  This *seems* to have done the trick, or at least it's made it past the trouble spot so far.

I'm not sure if it's appropriate to consider this "fixed" in the sense that it's corrected programatically, but I understand what's going on and a workaround at least.

ShanDestromp, 2020-11-12:
So after a lot of procrastination I finally got around to updating the 32u2 firmware with the linked repo and must report a success.  I've gone and tested known "failing" gcode and they've completed without issue.

To anyone else who might stumble upon this bug report:  Performing the fix isn't difficult, the most time consuming portion is gaining access to the Einsy to connect the pins, particularly if your printer is in an enclosure like mine.   The earlier "workaround" I mentioned (exporting your STL's with a lower poly count, which F360 calls "refinement") did work for me about 95% of the time (the remaining 5% was usually when using 3rd party STL's that I couldn't refine, and even then only semi-rarely).  On more complex shapes the difference was visible but only if doing a side-by-side comparison.  Basically if someone is encountering this issue they *should* plan on performing the firmware update, but save it for when you're already planning on doing printer maintenance, assuming you can live with reduced poly STLs.

closed 2020-11-12 by ShanDestromp