PrusaOwners/klipper #31: Internal error during ready callback: pat9125: Unable to locate stepper [extruder0]

ShanDestromp, 2019-12-18:
I'm guessing it's due to the recent upstream changes regarding extruder0/extruder naming rules.

printer.cfg uses current naming convention ([extruder] not [extruder0]) however pat9125 seems hard coded to expect extruder0.  I tried playing around with the python with no success; just kept kicking the error to a new line.

Short term commenting out relevant f-sensor lines in printer.cfg allows the printer to be functional.

I saw [your comment here](https://github.com/PrusaOwners/prusaowners/issues/65) that there were changes coming down the pipeline so you've probably already got this handled but if not, here's another one :-)
 
[klippy.log](https://github.com/PrusaOwners/klipper/files/3976244/klippy.log)

Arksine, 2019-12-18:
Thanks.  You are correct that its due to an upstream naming change.  Right now I don't have the pat9125 hooked up so I wasn't able to test it.  I will fix it and post it up shortly.

Arksine, 2019-12-18:
Should be good to go, let me know how it turns out.   If it doesn't work I'll hook up the pat9125 to a different printer (right now MMU2S is on the MK3) and get it resolved.

ShanDestromp, 2019-12-18:
Negative; and this is where I got lost when I was trying to fix it myself before:

> Traceback (most recent call last):
>   File "/home/pi/klipper/klippy/klippy.py", line 166, in _connect
>     cb()
>   File "/home/pi/klipper/klippy/extras/pat9125.py", line 207, in _handle_ready
>     self._init_stepper()
>   File "/home/pi/klipper/klippy/extras/pat9125.py", line 251, in _init_stepper
>     self.set_stepper(stepper)
>   File "/home/pi/klipper/klippy/extras/pat9125.py", line 355, in set_stepper
>     self.stepper_oid = stepper.mcu_stepper.get_oid()
> AttributeError: MCU_stepper instance has no attribute 'mcu_stepper'

Full log follows.
[klippy.log](https://github.com/PrusaOwners/klipper/files/3978464/klippy.log)

Arksine, 2019-12-18:
Ok, pushed another fix.  Hopefully should work ok now.

ShanDestromp, 2019-12-18:
Seems to have done the trick; querying the sensor works etc.

closed 2019-12-18 by ShanDestromp