PrusaOwners/klipper #32: Unknown config object 'heater'

brotherdust, 2020-05-14:
Hey! Hope you are doing well!
Noticed you pushed some updates 9 days ago and decided to update Klipper. Update process was without incident. However, when I try to control the printer, this shows up in `/tmp/klippy.log`:
```
   self.try_load_module(config, section_config.get_name())
  File "/home/pi/klipper/klippy/klippy.py", line 118, in try_load_module
    self.objects[section] = init_func(config.getsection(section))
  File "/home/pi/klipper/klippy/extras/probe_temp.py", line 268, in load_config
    return ProbeTemp(config)
  File "/home/pi/klipper/klippy/extras/probe_temp.py", line 53, in __init__
    self.sensor = self.printer.lookup_object('heater').setup_sensor(config)
  File "/home/pi/klipper/klippy/klippy.py", line 85, in lookup_object
    raise self.config_error("Unknown config object '%s'" % (name,))
Error: Unknown config object 'heater'
```
Here's my `printer.cfg`:
```
# This file contains pin mappings for Prusa i3 MK3 printers. To use
# this config, the firmware should be compiled for the AVR atmega2560.

# See the example.cfg file for a description of available parameters.

[probe]
pin: PB4
x_offset: 24
y_offset: 5
# This is for the old plate with tape on it
# z_offset: 1.2
# This is for THEKKIIINNNNG
z_offset: 1.6
#z_offset: 0.6
speed: 10.0
 
[bed_mesh]
speed: 300
mesh_min: 35,6
mesh_max: 240,198
probe_count: 7,7
move_check_distance: 5.0
mesh_pps: 1,1
algorithm: bicubic
split_delta_z: .02

[homing_override]
gcode:
 G1 Z3
 G28 X0 Y0
 G1 X101 Y100 F5000
 G28 Z0
axes: Z
set_position_x: 0
set_position_y: 0
set_position_z: 0

[gcode_macro G80]
gcode:
 G28
 BED_MESH_CALIBRATE
 G1 X0 Y0 Z0.4 F4000
 
[gcode_macro G81]
gcode:
 BED_MESH_OUTPUT

[stepper_x]
step_pin: PC0
dir_pin: !PL0
enable_pin: !PA7
step_distance: .01
endstop_pin: tmc2130_stepper_x:virtual_endstop
position_endstop: 0
position_max: 250
homing_speed: 50
homing_retract_dist: 0

[tmc2130 stepper_x]
cs_pin: PG0
microsteps: 16
interpolate: True
run_current: .281738
hold_current: .281738
homing_current: .149155
sense_resistor: 0.220
diag1_pin: !PK2
driver_IHOLDDELAY: 8
driver_TPOWERDOWN: 0
driver_TBL: 2
driver_TOFF: 3
driver_HEND: 1
driver_HSTRT: 5
driver_PWM_FREQ: 2
driver_PWM_GRAD: 2
driver_PWM_AMPL: 230
driver_PWM_AUTOSCALE: True
driver_SGT: 3

[stepper_y]
step_pin: PC1
dir_pin: PL1
enable_pin: !PA6
step_distance: .01
endstop_pin: tmc2130_stepper_y:virtual_endstop
position_endstop: -4
position_max: 210
position_min: -4
homing_speed: 50
homing_retract_dist: 0

[tmc2130 stepper_y]
cs_pin: PG2
microsteps: 16
interpolate: True
run_current: .3480291
hold_current: .3480291
homing_current: .182301
sense_resistor: 0.220
diag1_pin: !PK7
driver_IHOLDDELAY: 8
driver_TPOWERDOWN: 0
driver_TBL: 2
driver_TOFF: 3
driver_HEND: 1
driver_HSTRT: 5
driver_PWM_FREQ: 2
driver_PWM_GRAD: 2
driver_PWM_AMPL: 235
driver_PWM_AUTOSCALE: True
driver_SGT: 3

[stepper_z]
step_pin: PC2
dir_pin: !PL2
enable_pin: !PA5
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
#endstop_pin: tmc2130_stepper_z:virtual_endstop
position_max: 200
position_min: -2
homing_speed: 13.333

[tmc2130 stepper_z]
cs_pin: PK5
microsteps: 16
interpolate: True
run_current: .53033
hold_current: .53033
homing_current: .348029
sense_resistor: 0.220
diag1_pin: !PK6
driver_IHOLDDELAY: 8
driver_TPOWERDOWN: 0
driver_TBL: 2
driver_TOFF: 3
driver_HEND: 1
driver_HSTRT: 5
driver_PWM_FREQ: 2
driver_PWM_GRAD: 4
driver_PWM_AMPL: 200
driver_PWM_AUTOSCALE: True
driver_SGT: 4

[extruder]
step_pin: PC3
dir_pin: PL6
enable_pin: !PA4
# step_distance: 0.0035714
step_distance: 0.0010199485495570467063200705124
nozzle_diameter: 0.400
filament_diameter: 1.750
# The max extrude cross section is increased for the purge line, but in theory
# its probably a bad idea to have it this large, as its purpose is to catch 
# poorly sliced objects that extrude too much for small moves.
max_extrude_cross_section: 50.0
# Allows to load filament and purge up to 500mm
max_extrude_only_distance: 500.0
max_extrude_only_velocity: 120.0
max_extrude_only_accel: 1250.0
heater_pin: PE5
sensor_type: ATC Semitec 104GT-2
sensor_pin: PF0
#control: pid
#pid_Kp: 16.13
#pid_Ki: 1.1625
#pid_Kd: 56.23
min_temp: 0
max_temp: 305

[tmc2130 extruder]
cs_pin: PK4
microsteps: 16
interpolate: True
run_current: .513757
hold_current: .513757
homing_current: .314883
sense_resistor: 0.220
diag1_pin: !PK3
driver_IHOLDDELAY: 8
driver_TPOWERDOWN: 0
driver_TBL: 2
driver_TOFF: 3
driver_HEND: 1
driver_HSTRT: 5
driver_PWM_FREQ: 2
driver_PWM_GRAD: 4
driver_PWM_AMPL: 240
driver_PWM_AUTOSCALE: True
driver_SGT: 3

[heater_bed]
heater_pin: PG5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PF2
control: pid
pid_Kp: 126.13
pid_Ki: 4.3
pid_Kd: 924.76
min_temp: 0
max_temp: 125

[verify_heater heater_bed]
check_gain_time: 180
heating_gain: 1

[heater_fan nozzle_cooling_fan]
pin: PH5
heater: extruder
heater_temp: 50.0
fan_speed: 1.0

# Part Cooling Fan
[fan]
pin: PH3

[mcu]
serial: /dev/serial/by-id/usb-Prusa_Research__prusa3d.com__Original_Prusa_i3_MK3_CZPX1519X004XK20483-if00

[display]
lcd_type: hd44780
rs_pin: PD5
e_pin: PF7
d4_pin: PF5
d5_pin: PG4
d6_pin: PH7
d7_pin: PG3
encoder_pins: ^PJ1,^PJ2
click_pin: ^!PH6

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 1500
max_z_velocity: 10
max_z_accel: 200

# Keeps Debug LED off / not floating
[static_digital_output debug_led]
pins: !PB7

[output_pin BEEPER_pin]
pin: PH2
pwm: True
value: 0
shutdown_value:0
cycle_time: 0.001
scale: 1000

[gcode_macro M300]
default_parameter_S=1000
default_parameter_P=100
gcode:
 SET_PIN PIN=BEEPER_pin VALUE={S}
 G4 P{P}
 SET_PIN PIN=BEEPER_pin VALUE=0

[menu __filament __load]
type: command
name: Load Filament
gcode:
    LOAD_FILAMENT

[menu __filament __unload]
type: command
name: Unload Filament
gcode:
    UNLOAD_FILAMENT

[menu __filament __feed]
type: input
name: Feed Filament: {0:.1f}
parameter: 0
input_step: 1
gcode:
	M83
	G1 E{0:.1f} F400

# Firmware filament retraction. This enables G10 (retract) and G11
# (unretract) GCODE commands issued by many slicers. The parameters
# below provide startup defaults, although the values can be adjusted
# via the SET_RETRACTION command, allowing per-filament settings and
# runtime tuning.
[firmware_retraction]
retract_length: 0.5
#   The length of filament (in mm) to retract when G10 is activated, and to
#   unretract when G11 is activated (but see unretract_extra_length below).
#   The default is 0 mm.
retract_speed: 20
#   The speed of retraction, in mm/s. The default is 20 mm/s.
unretract_extra_length: 0
#   The length (in mm) of *additional* filament to add when unretracting.
unretract_speed: 7.5
#   The speed of unretraction, in mm/s. The default is 10 mm/s.

[probe_temp]
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PF3
t_offsets:
 35.0, 0.0
 40.0, 0.02
 45.0, 0.06
 50.0, 0.120
 55.0, 0.2
 60.0, 0.3

[tram_z]
[skew_correction]

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [extruder]
#*# control = pid
#*# pid_kp = 25.704
#*# pid_ki = 1.544
#*# pid_kd = 106.993
#*#
#*# [skew_correction 201912090954]
#*# xy_skew = -0.000285070201159
#*# xz_skew = 0.00056858455432
#*# yz_skew = -0.00491285890715
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	0.190000, 0.135000, 0.087500, 0.090000, 0.060000
#*# 	0.202500, 0.120000, 0.065000, 0.100000, 0.025000
#*# 	0.185000, 0.067500, 0.005000, 0.017500, -0.045000
#*# 	0.157500, 0.070000, 0.020000, 0.040000, -0.055000
#*# 	0.187500, 0.057500, -0.012500, 0.012500, -0.045000
#*# x_count = 5
#*# y_count = 5
#*# mesh_x_pps = 5
#*# mesh_y_pps = 5
#*# algo = bicubic
#*# tension = 0.2
#*# min_x = 35.0
#*# max_x = 240.0
#*# min_y = 7.00000000002
#*# max_y = 198.0
#*#

```
My best guess is that I don't have a specific `heater` section, but I'm not familiar with the internals of Klipper enough (yet).

Please advise. And THANK YOU for doing what you do!

Arksine, 2020-05-14:
It was a bug due to an upstream internal API change.  It should be fixed now.

brotherdust, 2020-05-14:
Nice work! Thank you so much!

closed 2020-05-14 by brotherdust